stuartpittaway/diyBMS-CurrentShunt #1: add an option form a 300A 50mV Shunt

chapulino, 2021-06-17:
It would be interesting to have an intermediate option between the 200A and 500A Shunts.
That is why I suggest that the selection of a 300A sunt be considered, which would be an intermediate point

SergioRius, 2021-06-18:
I would second this. As the recommended fuse by Victron in their Multiplus2 5000 is a 200A one, someone would expect a 200A shunt to be fairly tight and depending on the provider and quality, not enough.
Is this range simply programmed in the firmware or it depends on a boards component?

stuartpittaway, 2021-06-18:
Purely software controlled, so happy to add the other options

SergioRius, 2021-06-18:
> Purely software controlled, so happy to add the other options

We've been looking at the code this afternoon and saw that it nicely accepts the shunt capacity on the 18th address.

And the following link as been moving around as an cheap supply of shunts for testing. The nice thing is that it has a selection of device capacity:

https://a.aliexpress.com/_vSp8LH