stuartpittaway/diyBMS-CurrentShunt #2: PCB does not fit 200A shunt

magpern, 2021-07-03:
The 200A shunt does not fit on the pcb.
The holes on the shunt is of by about half a hole.
And the shunt covers the pcb corner holes.

This is the shunt, I think it is the correct one.
https://www.digikey.se/short/bd507d49

[image]
[image]
[image]

stuartpittaway, 2021-07-04:
Ok, I'll update the PCB. There are no components/traces on that side of the board, so you should be safe to drill a new hole.

stuartpittaway, 2021-07-06:
Can I ask you to confirm these measurements against the real shunt you have?

It should be 82.5mm by 44.4mm

[image]

magpern, 2021-07-06:
I finally found my slide measure tool. My son tidied the garage.. garage looks nice.. tools are a little random.

The outer dimensions of the actual shunt (200A) is pretty must correct.
But the hole placement is not.
I used a digital slide measure and got the holes to 5.3mm, which is also proven by the other measurements, hole-side to hole-side, and wall to hole. I measured several times and got almost the same numbers.

A = 3.7 mm
B = 26.6 mm
C = 37.2 mm
 
[image]

stuartpittaway, 2021-07-07:
Thanks - as I thought the physical device doesn't match the data sheet!

I've modified the PCB to add another 5mm to the slot so it should now fit.  I'm doubtful that it will still fit inside the green DIN rail mount now though.

closed 2021-07-07 by stuartpittaway