stuartpittaway/diyBMS-CurrentShunt #4: Over/under current limit not settable

magpern, 2021-09-02:
My system can pass over 180 amps. But I cannot set the over/under current limits to more than 127, and if I do, they become -127.
127+127 = 0xFF = one byte. Is the field too small?

[image]

stuartpittaway, 2021-09-02:
What are the shunt amps and MV rating? That determines the maximum.

magpern, 2021-09-02:
> What are the shunt amps and MV rating? That determines the maximum.
> 
> Those settings only control the relay on the shunt itself, not the controller.

[image]

stuartpittaway, 2021-09-02:
Ok, the maximum should be higher than that. I'll take a look at the code.

The max should be

Total shunt amp rating * (40.96/50mV)

stuartpittaway, 2021-09-03:
Ok, in which case this is correct.

This is how the INA228 chip works.  given you are trying to use 180amps through a 200amp shunt, I believe the shunt is under rated.

You should have a shunt with at least 25% overhead - or about 225amps (probably 250amp shunt would have been better).

magpern, 2021-09-03:
Highly unlikely that I personally would use 163 amps, I will be around 130 amps max. But regardless. If you type 180 in the field, it accepts it, but changes it to some negative number instead.

stuartpittaway, 2021-09-03:
Ok, the controller doesn't calculate what is a correct or incorrect value, its simply passing it to the code on the current monitor board, and that then accepts/rejects the value and applies a default if its out of range.

I probably need to make that a little clearer on the screen.

The negative/positive numbers relate to the different limits you can have for charge or discharge.

The "Over current limit" is positive (used for discharge of the battery), the "Under current limit" will be negative (used for charging the battery).

Normally most people will set these to be the same - or just ignore that section completely if they are not using the relay on the current monitor board.