stuartpittaway/diyBMS-CurrentShunt #5: SoC not resetting

magpern, 2021-09-03:
I have 55V set as fully charged voltage. And tail to 3 amps. It hit 55V, stayed around this voltage for 1 hour. Now inverter is going into "Float" at 54.50V. SoC is still 105%

[image]

[image]

And as an extra feature. A field to manually set a presumed SoC

stuartpittaway, 2021-09-03:
Ok, the current is negative which means you are discharging from the battery.

The reset only occurs when the charging "tail current" (positive value) is reached.  So you would need the voltage to go over 55V (at the shunt voltage) and the charging current to be below 3amps for over 3 minutes.

magpern, 2021-09-03:
At 10:53 it hit 55V (but setting was 55.50V)
I changed it to 55V at this time.

It was arount 55V +-50mV for close to an hour, and was below 5 amps several times.
I think it could be the changing of the voltage value from 55.5V to 55V, while it was above 55V.

Attached is the voltage curve. Red arrow is when it hit 55V.

I also think 5 amps is too low for me, I will change that. In Victron you don't specify the amp, you set a percent value. I think I have 2 or 3% in my SmartShunt. Ant it is a percentage of the Ah of the battery. So I'll bump it up 10 or 15 amps.

[image]

stuartpittaway, 2021-09-03:
Might be worth logging the shunt monitor values from the diyBMS so you can see what it thinks is going on.

magpern, 2021-09-03:
> Might be worth logging the shunt monitor values from the diyBMS so you can see what it thinks is going on.

I will do that and report back. I have an influxdb in home assistant. I will try and get diyBMS to report to it

chapulino, 2021-11-07:
Magpern,  could you solve it, am I in a similar situation? Although in my case the parameters are a little different because it is a larger battery (600A) so the tail curren is at 12A (Victron recommends for this value in lifepo4 batteries, 2% of the battery capacity)  
[image]
[image]

chapulino, 2021-11-08:
I finally got it to synchronize at 100% it was silly, it had a maximum voltage value, now in all the places it is 54.4V and it is going great. Never with Victron have I had a 100% load so stable and with so little deviation in the cell voltage. As the maximum approaches, the current is gradually reduced until it reaches 100% load, the operation in my opinion is impeccable,
TIP I have no difference between maximum voltage and float (I think that step is reminiscent of lead batteries