stuartpittaway/diyBMS-CurrentShunt #7: Sourcing parts for CurrentShunt and Victron Integration

allanlind, 2022-03-21:
Hi Stuart, I have trouble sourcing some of the parts for the CurrentShunt and wondered if you might have ideas where to buy.
ADM2483BRWZ, ATTINY1614-SSNR, LM5009AMMX/NOPB, INA228AIDGSR are not available on Mouser or Digikey.
The ADM and ATTINY I managed to find on Ali Express or Ebay. But the LM5009 and INA228 are a problem.
I only need enough parts to build 2 current shunt boards. Do you by chance have any extras I can buy?

The reason for building the shunt is I cannot get the diyBMS Controller to work properly with the Victron GX inverter. I was hoping the problem would go away if I added the CurrentShunt to report the current flow in/out of the battery bank. Somehow the Victron is not working as it should in ESS mode.
Trying to keep grid at 0 is impossible as Clouds come over. And you can see the battery is not showing Current and charge level. 
[image]

 I have Victron DVCC enabled so Victron gets its voltage from diyBMS controller but I have SCS (Shared Current Sense) turned off. 
[image]

What happens is I have to manually adjust Grid Setpoint +/- 600W depending on house load and solar output. 
[image]

I asked Victron for support and they said they will not support me as I am using an unsupported battery. 

I figured if I have the CurrentShunt connected I can enable SCS.

stuartpittaway, 2022-03-22:
Hi.  Unfortunately I don't have any spares for the current shunt - as you mention, the parts are almost impossible to obtain at the moment.

I don't have much experience with Victron equipment - but I don't understand why you need to change the grid setpoint - I doubt DIYBMS would have any effect on this parameter.

Are you missing a current shunt/clamp to limit import/export ?  

How can the Victron monitor the grid to determine if it needs to power the loads or not?

allanlind, 2022-03-22:
The Victron Multiplus has AC current monitoring on it’s input and output so i don’t see why it would need a shunt on AC. It should also be measuring the DC current into/out of battery. ESS is a Victron script you load in to the Multiplus which turns it into a smart energy storage system. Optimising battery life and minimising grid usage. 
I am not sure how to explain this well. But all I want is to keep the grid set point at zero, so the Victron never uses power from the grid. Only feeds surplus back in when battery is fully charged. But the Victron won’t do this, it’s using power from the grid when my solar panels are working. Victron refuse to support this. The only way I can get the grid usage under control is to set the Grid Setpoint at -600W. But as the solar panels output varies on cloudy days the grid draw varies all over fhe show. Victron won’t support the inverter as I am using an unsupported battery. As I don’t have a supported battery I cannot troubleshoot. The only thing I can think is that I need to have shared current sense turned on and have diyBMS instruct the Victron the battery DC current. 
Does this make sense?

stuartpittaway, 2022-03-22:
The current shunt on diybms will only tell you whats going in and out of the battery, but that won't give you the load on the house/grid.

allanlind, 2022-03-22:
I understand that. What I am trying to say is that the current shunt will tell diyBMS controller the instantaneous battery current , the +/-W setting and the battery charge %. The diyBMS controller will then pass these parameters on to the Victron via canbus. My guess is the Victron ESS script must have these readings for it to calculate grid current usage. Battery status may be critical for ESS to calculate whether to use power from the grid. It might think the battery is in an unhealthy state and needs to urgently use grid power. Just my guess. If you scroll up you will see my screen shot of the Victon console which is not displaying the Battery Watts, Amps, and % Charge. I think this is the problem. For some reason If I unplug the diyBMS controller from the canbus the Victron console shows the battery status as 100%. But then ESS goes into bypass mode.

tmichel-gh, 2023-02-22:
Hey Stuart,

if you're planning a redesign of the CurrentShunt board, what about switching from LM5009A Buck converter to an isolated 5VDC-5VDC-Converter and driving it through the 5V provided by the controller board (J5 Pin 1)? Also decreases the small energy amount consumed by the CurrentShunt itself, because only the measurement-voltage needs to be driven by the battery