stuartpittaway/diyBMS-CurrentShunt #8: False current readings

red0909, 2022-07-23:
Hi, now i have detailed information about this issue.

so first 
i have a 200A 60mV shunt 
in discharge mode current readings are correct to -20mV shunt voltage below this voltage the current readings start to decrease and rise again randomly.

in charge mode this problem occures at 30mV over this voltage the current readings drop and rise again randomly.

additional the mV readings start to be inacurate for more than 30mV in charge and less than -20mV in discharge mode
if i stay within this limits all works fine.

you can use a small shunt like 50A 60mv to test this

thanks for your work.

stuartpittaway, 2022-07-24:
The INA228 chip has a scale of ±40.96 mV

If you are using a 200amp shunt, and driving 200A through it, it will be inaccurate, as the ADC will be saturated.

red0909, 2022-07-24:
does it mean -40 to 40mV or -20 to 20mV scale? 
i thought the first.

stuartpittaway, 2022-07-24:
Yes the first.

red0909, 2022-07-24:
but the -40 to 40 does not work, maybe a hardware problem on my shunt pcb? 
have you tested this range?

stuartpittaway, 2022-07-25:
How are you reading the mV value? Is this the value shown in the web page?

red0909, 2022-07-25:
i am reading it with a precise voltage meter. the readings correspond with the web page readings, if i stay in the range of 20 to -30 mV
i have 16s lifepo. at the moment limited to 100A charge and 65A discharge, i am waiting for a new shunt to be able to get 300A readings in future. 
so if you can, please test this range in your shunt you can use a smaller one if you dont have the load for test.

stuartpittaway, 2023-01-25:
Hi @red0909, did you resolve this issue in the end?

Its probably worth updating the code on the current shunt as the SoC calculations have changed, along with the amp-hour counters.

I don't have a 60mV shunt to test what you are seeing in the previous tests.

red0909, 2023-01-25:
i have solved this issue with a bigger shunt 500a 50mv so that i stay in this -30/20mv range.
but i can test this new code with the 200a shunt again, i will let you know.