stuartpittaway/diyBMS-CurrentShunt #11: one shunt per bank ? Enhancement

N1c084, 2022-08-20:
Hi,
Any idea to drive (without too much work) 1 shunt per bank?

Maybe the simple way is to have 1 contrôler per bank, but there is an easy way to have one master controler to obtain all data in one Web UI?

Thanks

stuartpittaway, 2022-08-20:
Technically it's just reading more MODBUS devices (protocol the shunt uses). The controller code would need significant changes to support this as well

N1c084, 2022-08-20:
yes i know for modbus communication, but i try to find a simple way to handle it, if possible.
the idea of a master controler is also a big code chalange?

stuartpittaway, 2022-08-20:
Master controller is a much bigger change

N1c084, 2022-08-20:
Ok.
Please can you globally detail the step required to add multiple modbus
shunt reading and reporting to web Ui ?
I will try some experience to do it