stuartpittaway/diyBMS-CurrentShunt #15: Help programming new latest code into the current shunt.

anthonyevs, 2023-01-27:
I have a problem with the current shunt. It came to me pre programmed, and the issue is the shunt wont accept any new setup for my battery, ie, in basic settings its not saving my settings.
Stuart has updated the code and is ready to download via github.
I havent got an UPDI programmer, thinking i should buy one, or convert my arduino UNO into one
I have followed steps to convert the arduino into a jtag2updi interface.
The sketch i loaded was from Eltangas and i have followed the steps to upload this sketch onto the arduino.
Can anyone give a method / instruction to get the arduino to program the current shunt with the latest code....
or,
give advice on which UPDI programmer i should buy, and how expensive are they,
Im assuming the UPDI programming is straight forward, so i`d lean that way for the expense.  
Any help greatly appreciated in getting the latest software code uploaded onto the current shunt,
Thank you,
Anthony.

stuartpittaway, 2023-01-27:
Hi.  You no longer need an UPDI programmer or old Arduino to program UPDI style chips.

Take a look at this [UPDI Programming](https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md#wiring-the-hardware)

You can use a simple USB to Serial adapter instead.

stuartpittaway, 2023-01-28:
Once you turn the Arduino into an UPDI programmer, you then need to use a program on the computer like pyupdi, or jtag2updi (both are Python programs).

This then talks to the Arduino, and in turn to the ATTINY chip connected to that - you "upload" the DIYBMS code from your computer - it's not put onto the Arduino.

deekoowee, 2023-02-05:
Hi, 
I have a similar problem, when trying to save basic settings for the current shunt . Controller start the saving procedure, but returns all back to zero (maybe, because the settings doesn't store into 1614 EEPROM ?). Controller and shunt have latest firmware, tried also the Pylontech firmware (which is my goal) on controller without success. Have tried 19200 8N1 and 9600 8N (changed the dip on shunt). Pylontech firmware give RS485 error on display.

1614 programming Arduino UPDI seems to be ok with Avrdude 7.1. Writing and verifying is ok.
Bootup shows continuous red and green 8sec on and then red blinks 5 times. 
After red, the green blinks 6 times. 
When green stops blinking red blinks once and then 5sec pause
After 5sec pause, red blinks twice with 2sec pause in between then 5sec pause, this happens 4 times.
In fourth time, the red blink only once and bootup starts at beginning.

Tried to figure out the error codes (two blinks means configuration error), but I'm not sure of red first and last single blink.
Checked the INA, 1614 and ADM solderings and tested the conductions, no problems.
Replaced the ADM chip twice and 1614 once, but no help. 

Controller works with PZEM, so controllers in not the issue.

I have tested the serial connection with logic analyzer. It shows, that RX is getting some data from controller. TX from 1614 doesn't send anything and XDIR never change state. 

What can I check next ?

stuartpittaway, 2023-02-06:
Hello.

If you are getting a regular "red led" flashing every 2 seconds - that's a good sign, communication between the INA and ATTINY chip is working.

So I would expect there to be an issue on the soldering between the RS485 isolator chip and the ATTINY.

The pins are small, so you need a microscope or magnifying glass to confirm no solder joints/bridges.

stuartpittaway, 2023-02-06:
You should see comms from the controller every 4 seconds to the shunt over RS485, thats how frequently it polls.  The protocol is MODBUS if your scope can decode it.

If the shunt receives a valid command, it will flash the green LED.

Can you check with the scope on the actual RX/TX pins on the ATTINY chip?

Additionally, are you using the **pre-compiled** ATTINY code from GITHUB, and programming with avrdude ?

Have you tried swapping A and B connections on one end of the cable, just to check ?

deekoowee, 2023-02-06:
Analyzer can decode MODBUS. This is the 4ms data from controller;
[image]
Very strange, that the slave ID = 169

No other green flash for valid command.
 
Actual RX/TX ... Do you mean RX/TX at pinheader ?
I have used those pinheader pins.

Yes, I have used the pre-compiled code from Github and avrdude. Arduino Nano UPDI USB programmer is made with these instructions; [https://daumemo.com/diy-updi-usb-programmer-which-can-be-made-with-cheap-hardware/]

Swapped A and B, but no help.

stuartpittaway, 2023-02-06:
Did you change the modbus address in the controller after using the pzem? Should be 0x90 for diybms

stuartpittaway, 2023-02-07:
Also, check the baud rate is back at 19200 like the screenshot below

[image]

deekoowee, 2023-02-07:
Modbus address is 90 and baudrate 19200 8N1

stuartpittaway, 2023-02-07:
Ok, have you tried swapping A+B connections on one end of the cable?  Double check the 120ohm resistors are on both ends of the line (are you using wire resistors, or the on-board SMT parts?)

The ATTINY isn't going to respond until it gets a good MODBUS signal - which should be visible on the TX/RX pins.

deekoowee, 2023-02-07:
Yes, I have tried swapping A+B on one end.
Using on-board resistors. JP4 soldered at controller board.

stuartpittaway, 2023-02-07:
Did you also solder the resistor on the shunt side?

deekoowee, 2023-02-07:
R12 is soldered at shunt and JP1 is bridged as default.

stuartpittaway, 2023-02-07:
Ok, I'm running out of ideas here, obviously until the attiny receives a valid modbus message, it won't reply.

Have you installed the dip switches? These should be all off for standard use.

deekoowee, 2023-02-07:
Dip switches are on board and they are off.

deekoowee, 2023-02-07:
I connect the analyzer directly to controller's ESP-module pins IO21 and IO22.
The data is similar as my earlier data. The controller seems to send data to modbus address 169.
Controller web has configured for address 90.
I'm not good enough with software. 
Just thinking, could it be possible, that controller has not fully overwrited the modbus address after PZEM test ?

stuartpittaway, 2023-02-07:
It won't hurt to fully flash the esp32 again (select wipe flash).

If you have a scope, also check the i2c pins on the shunt to see traffic to the ina228

deekoowee, 2023-02-08:
Wiped the flash and reprogrammed Esp32.
Same issues continue. Shunt settings would not be saved at controller settings and modbus shows address 169 at controllers IO21/IO22 and shunt RX/TX pinheader.

I2C have some traffic on SDA ans SCL

stuartpittaway, 2023-02-08:
Can the scope decode the RS485 from A/B connection ?

I can't see anywhere in the code which would force 169 modbus address.  Does the scope support little endian format?

stuartpittaway, 2023-02-08:
Another thought - if you swap back to using the PZEM device, does the scope read the correct modbus address from that? (address should be 1)

deekoowee, 2023-02-08:
Sorry, my fault of using the analyzer. Having the LSB first and got;
[image]
Now, the address shows 90.
Around 2,8s after RX, TX have one pulse, but no data.

stuartpittaway, 2023-02-08:
Great, does the green led flash on the shunt when that data arrives?

deekoowee, 2023-02-08:
No other green flash, except long green + 6 flashes at boot sequence

stuartpittaway, 2023-02-08:
Ok, just checking it's an attiny1614 chip you installed.

You see the modbus signal on the Rx pin on the shunt?

deekoowee, 2023-02-08:
Yes, it is attiny1614. 
Yes, modbus signal is from shunt RX pin

stuartpittaway, 2023-02-08:
Ok, it could be something like the fuses on the attiny.

Might be worth reprogramming that again over updi.

And I'd also run a soldering iron around the attiny pins again, just to double check.

Is it receiving 5V ok?

deekoowee, 2023-02-08:
Attiny pins soldered again and checked with microscope.
Reprogrammed 1614 with UPDI.
5V power is ok.

stuartpittaway, 2023-02-08:
I don't think there is any power saving code in the shunt, so I won't be disabling the usart device.

I'm struggling to think what else it could be.  Could it be a fault on the tiny chip?

deekoowee, 2023-02-09:
Attiny chip has replaced to new one. Old attiny was working / not working similar way.
I think, the main problem is the communication between controller and shunt.
Shunt get the data from controller, but do not send anything back (except broadcast).
Could there be a difference at attiny chip releases ?

stuartpittaway, 2023-02-09:
If the attiny was receiving valid communication it would flash the green led. I'll have to take a look at the code to check.

deekoowee, 2023-02-09:
[Release-2023-01-25-12-26](https://github.com/stuartpittaway/diyBMS-CurrentShunt/releases/tag/Tag-2023-01-25-12-26) code in use.
1. At boot, the red blinks 5 times with 200ms/200ms cycle.
2. After red, the green blinks 6 times 50ms/150ms cycle.
3. When green stops blinking red blinks once 800ms and then 5sec pause.
4. After 5sec pause, red blinks twice 800ms with 2sec pause in between then 5sec pause, this happens 4 times.
5. In fourth time, the red blink only once 800ms and bootup starts at beginning after 2s pause.
6. Next both lights for 8s
Then back to 1.

stuartpittaway, 2023-02-09:
Ok, the reboot loop is indicating a problem with i2c communication to the ina228 chip.

If you are getting a single red flash - thats normally bad comms.

```
// Sequences used to indicate error on RED led
// Single flash
const static uint32_t err_INA228Missing = 0xF000000F;
// Two flashes
const static uint32_t err_InitialConfigure = 0xF0F0000F;
// 3 flashes
const static uint32_t err_WrongChip = 0xF0F0F00F;
// 4 flashes (0xCC=2 flashes)
const static uint32_t err_WriteConfig = 0xCCCC000F;
// 5 flashes (0xC0=1 short flash)
const static uint32_t err_WriteADCConfig = 0xCCCCC00F;
// 6 flashes
const static uint32_t err_WriteRegisters = 0xCCCCCC0F;
// 7 flashes (0xAA=4 fast flashes)
const static uint32_t err_INA228Reset = 0xAAA8000F;
// 8 flashes  10101010101010100000000000001111
const static uint32_t err_CheckSumErr = 0xAAAA000F;
// 9 flashes 00101010101010101010000000001111
const static uint32_t err_ResetChargeEnergyRegisters = 0x2AAAA00F;
```

Can you put the scope on the i2c lines and see what reply you get from the ina228?

deekoowee, 2023-02-09:
I was thinking, that those single red flashes are meaning beginning and end of error sequence.
There's 3 double flashes between them.
INA is soldered correct way and solders were checked with microscope.
Because of difficulty of INA availability, I don't have spare.

I can get only undecoded waveform from I2C. Following is triggered from SCL =0 to 40us
[image]

LM5009A is in use and 5V is available.

stuartpittaway, 2023-02-09:
The "WrongChip" 3 flashes code is commented out, so that won't be the cause.

The "missing INA chip" (no answer) flashing light pattern is 0xF000000F
So that should be 1600ms RED ON, no lights for 9600ms, 1600ms RED ON - although the pattern you see seems to indicate shorter times than that.

Then a 2 second delay, before repeating 3 times.

After this, the shunt will timeout and reboot.

deekoowee, 2023-02-10:
Two flashes = err_InitialConfigure and it is correct, because saving of the configuration do not work.

I look controller code and compared it to analyzer data. When I click at Current & Voltage Monitoring page 'Save basic settings', this data has taken from esp32 IO21/IO22. Settings wouldn't be saved like **anthonyevs** has reported at first message of this thread. Just same data arrived to shunt.

I'm not good with this, but do I underestand from code, that first register should be 40019 ? However, analyzer data shows 30019. 
I can't find 30019 from [MODBUS registers](https://github.com/stuartpittaway/diyBMS-CurrentShunt/blob/master/Code/diybmsCurrentShunt/MODBUS%20Registers.md)
'number of registers to write' seems to be correct.
Function code has also some mystery (10 in data vs 16 in code)

How the web browser handles the data, when clicking the 'Save basic settings' ?

[image]

stuartpittaway, 2023-02-10:
If you are getting flashing led and a boot loop, non of the rs485 Comms will be working, the attiny will ignore all of that.

deekoowee, 2023-02-10:
So, the problem should be at Attiny.
Data from controller is available at Attiny pins 6 and 7, but Attiny do not accept any data.
I'll replace third Attiny and see what happens.

stuartpittaway, 2023-02-10:
No, the code on the attiny is running, but as it cannot communicate with the ina228 so it aborts and reboots.

It will not answer or listen to the serial RS485 data whilst in this state.

anthonyevs, 2023-02-11:
keeping tabs with this thread, if i have the same problem as deekoowee, i hope he can work thro it tp solve it, as he has more knowledge, expertise and equipment to test than i do.
I`m still at the "trying to update the software program stage lol...." have now bought a usb ttl adaptor and in the process of reading the thread on how to program using it.
Im really needing t gain knowledge on this stuff. Im a good engineer mechanically, have good electrical skills etc.... but electronics and programming is something i think i need to learn alittle about. Im god at plc ladder too, but that seems yesteryear tech these days.
Trying to find a local youngster with coding skills to pay him for a short crash course in the basics.
Im gong to need a basic skill level going forward. There are so many home automation stuff that the very cheap aduino can do these days.

deekoowee, 2023-02-11:
I ordered [MIKROE-4810](https://www.mouser.fi/ProductDetail/Mikroe/MIKROE-4810?qs=A6eO%252BMLsxmSz%252BFqnPV7jlA%3D%3D) from Mouser. 
It is more expensive than native INA228 chip, but atleast I can have it.
I will connect it with jumperwires to shunt. Have to adjust the address shorts on board and connect Alert wire to Attiny pin 8.

Well, next week I have more info about this.

deekoowee, 2023-02-15:
I confirm, that MIKROE-4810 works and shunt started to operate.

Just thinking of the error indicating flash of red led.
On my case, first single flash was giving hint of INA error, but it continued with two flash sequences and last was also single flash before rebooting occured.
I think, with two errors, the flashing should be 4 times ; one short flash - pause - two short flash - pause eg. INA error - pause - config error.
Earlier saved parameres were shown at config after changing the INA. Strange, because shunt give two flash error code.

Ok, now shunt shows 4,02A current / 215W power / 5.04Ah in without of any load or charger connected to battery.
Only the controller and shunt are connected to battery and measured total current consumption is 35mA
I have 14S Leaf bank. 

Should I have some different shunt firmware ?
What I have learned from source code, the shunt firmware is more for 16S LiFepo

stuartpittaway, 2023-02-15:
The shunt code is completely generic, but it does default to lifepo4 settings, which can be changed in the interface.

What shunt are you using ? MV rating/current rating.

Have you fitted the shunt in the positive cable? (You should)

Is the vsense wire connected?

deekoowee, 2023-02-16:
Shunt is 500A/50mV, connected to positive cable and vsense connected to battery positive terminal.
I tried to change calibration settings to 32 and I got this;
[image]
Is there rules to calculate the calibration value ?

stuartpittaway, 2023-02-16:
Hi @deekoowee, there shouldn't need to be any calibration done as long as you have the correct shunt size selected in the configuration page.

Ignore the "Ah In/Out" values for now - can you apply a consistent known current on the battery to see if the current reading changes?

With a 500amp shunt, it won't be very accurate at low currents (probably less than 1amp) so ideally try and charge the battery and compare to the current reading on the charger.

stuartpittaway, 2023-02-17:
As you have seen, the shunt should automatically calculate the correct calibration value - based on shunt current amps/mv scale - it does this when you hit the "save basic settings" button.

Afterwards, you can override the calibration value using the "save advanced settings".

I've run the settings you have through the TI configuration page, (https://dev.ti.com/gallery/view/4910879/INA228_229_237_238_239EVM_GUI/ver/2.0.0/) and it looks like its a valid set of values.
[image]

You should be able to read up to 409A on the shunt you currently have.

Have you removed the INA228 chip from the MIKROE device?  or are you trying to use it in place?

deekoowee, 2023-02-17:
What does the CURRENT_LSB mean ?

I use MIKROE, until I receive the INA chips.

Strange, that S.o.C is now 123,10% (increasing all the time) and Current is 8,44A.
Does the values be calculated somewhere in code ?

stuartpittaway, 2023-02-17:
Current lab is the smallest value the current shunt can read, in bits.....

If the current reading is not zero, then the SoC will obviously change.

The mikroe isn't the same circuit as diybms, it's missing the filters on the inputs.

deekoowee, 2023-02-17:
OK, I'll solder two jumper cables to C14, then connecting the cables to mikroe and connect the shunt resistor to diybms.

deekoowee, 2023-02-17:
Above was bad test. I read the INA datasheet. There was mentioned that noise to wires will cause error to measuring.

I decided to desolder INA from mikroe and solder it to diybms.
Using very short wires to shunt resistor, current display start to give reasonable values.
Charger disconnected, current display shows 0,04A charge. 

Much better now, thanks for pointing to diybms filter circuit.

stuartpittaway, 2023-02-20:
@anthonyevs is this issue resolved now?

anthonyevs, 2023-02-20:
Hi Stuart, i havent had an opprtunity to look at the shunt issue, and was waiting to see if deekoowee resolved his issue as he seems to hav the same issue as i had.  PlusI have been having various other issues of my own making in the short term when i sadly swapped the tx/rx connectors after a loading of the new latest pylontech emu code onto the controller.
I blew the esp32 only, no other parts on the controller were damaged by the looks of it, as a new esp32 was fitted and the controller appears to be working fine now, 
However, sadly, it appears that i have damaged ALL 16 of the v4.4 cell boards. I have set up a small test rig with an older controller and single battery to test every cell board individually, and non of the cell boards sends or receives any packets of data. Got to try to find a test method to test the bad parts on the cell boards to get them repaired with new parts.
Thanks for checking in on me thought Stuart, much appreciated.

stuartpittaway, 2023-02-20:
Hi @anthonyevs , I don't know how you can damage the cell modules - they are all electrically isolated from each other through an optocoupler.

When you tested them, did you ensure you were running at the correct baud rate/communication speed?

anthonyevs, 2023-02-20:
Hi Stuart,
all i can write is what happened. All those 16 cell boards had been connected to the v4.2 controller and running without any dropped packets for over 2 weeks, working perfectly.
then i unplugged all the cabling out of the conroller mainboard (shunt, relays +5v input, rx / tx out of the mainboard to do the software update on the esp32 to the latest version.
After successfully updating the controller software, plugged back in all cables, then obviously noticed the cell boards were not syncing with the controller, noticed i had reversed the tx/rx at the controller. 
Changed the controller for a spare that i had here for another system, still the cell boards would not sync, not receiving packets. 
Swapped them out for spares i have (for the other system im building) and now all of the replaced cell boards are working, i havent changed any other settings, baud rate, nothing.
When testing the cell boards i removed, on a small test rig with the older v4 controller, when the power connector is plugged into the "damaged" cell board (no tx/rx connected yet) the blue led flashes twice, then twice again after a short pause, then the bleed led lights solid red, and the cell board starts bleeding from the battery, plug in tx/rx, and nothing, the controller will not recognise it.
Is it worth reprogramming them?
Or can you advice anything else to check for.
Thanks

stuartpittaway, 2023-02-20:
I would try reprogramming them, I bet they come back to life.

anthonyevs, 2023-02-21:
Hi Stuart, reflashed all the cell boards, and YAY, they did all come back to life.
Again, thank you.
On a seperate topic, im going to try to connect the BMS to the inverter today via the canbus. I`ve read thro the pin out on the canbus cable on my solis inverter, is seems to be pin 4 (blue) Can-H, pin 5 (blue/white) Can-L, pin 6 GND. Does that match your pinout on your sofar solar set up also Stuart, as most chinese stuff is made to similar spec.
Thanks

stuartpittaway, 2023-02-21:
Each inverter is quite specific on the wiring.

Try connecting it up without the ground to start with and see if it works - this can help to avoid ground loops.

anthonyevs, 2023-02-21:
Thanks Stuart, Ive connected as stated, blue is Can-H, blue/white is Can-L, set the bms to pylontech protocol and enabled, the inverter has seen it, and i get the CAN ok message on the inverter so it is working. 
Now, the bms needs the current shunt to work, as my battery is currently at 83% (measured by a seperate shunt, a junctek VA monitor) I am using this as an extreme out of operating conditions emergency stop control method, to trip the relay / contactors if the bms goes awol or the inverter goes awol, then this will act as an out of condition final emergency stop, probably overkill as a 3rd control measure, but i dont mind.

anthonyevs, 2023-02-21:
Hi Stusrt, please wee the pic of the shunt set up page. It will not save the fully charged value, tail current or efficiency. Also see the adv settings values.
[image]

anthonyevs, 2023-02-21:
Also, to answer one of your questions, the voltage is reading pretty much the same as the bms, its reading 0.2v higher

deekoowee, 2023-02-21:
The remaining 0,04A charging error is solved also and current reading with disconnected charger is zero now. 
I found fix from INA datasheet. I connected the shunt module +V SENSE to shunt resistor load side of shunt resistor and shunt module +++ to battery side of shunt resistor.
I have charged the bank for several days with small current. Diybms charging current reading differs only 0,02A at 1A charging current, comparing to multimeter reading.
Difference was same with 4A charging current.

Should I reset the S.o.C. % after the bank has reached the top voltage ?

[6 more replies]

closed 2023-02-27 by stuartpittaway