stuartpittaway/diyBMS-CurrentShunt #16: Current reading not correct

stefanK74, 2023-02-17:
The shunt board (500A/50mV) seems to be electrically working, but current readings are not correct.
The displayed current/power values also do not correlate to the amps going in or out.
The shunt´s voltage reading seems to be ok.

Upon powering up the shunt the green/red leds flash: 5xgreen, 1x red, then periodicaly green and red
Shunt is installed in the positive cable.
Vsense wire is connected to the battery.

Platform & Version
Host name: DIYBMS-009C7938
Processor: ESP32
Version: 7eebc8252a0a6f83da919580a9e9534cacd37f4d
Compiled: 2023-02-14T09:29:32.618Z
Language: en
SDK Version: v4.4.3
Min free Heap: 87912
Free heap: 99280
Heap size: 283788

Any help would be highly appreciated!
Thanks!

At this point in time the batteries were charged with about 47 Amps.
mqtt: {"valid":1,"voltage":54.029,"current":-0.0639,"power":3.5919,"mAhIn":0,"mAhOut":10,"temperature":24,"relayState":0,"soc":89.49}

[image]

stuartpittaway, 2023-02-17:
Ok, assume the shunt voltage @ 54.02V is correct.

Can you confirm that the shunt is installed in the POSITIVE cable - AFTER all the 4 banks merge together.

You then also have the two sensor wires connected to the shunt (normally screw terminals) directly to the shunt PCB (normally soldered to board).

Additionally, you have two power feeds going directly from the battery (after positive merged together) to both "+++" and "+VSENSE" terminals ?

A very crude diagram...

[image]

stefanK74, 2023-02-17:
>Can you confirm that the shunt is installed in the POSITIVE cable - AFTER all the 4 banks merge together.
Yes.

>You then also have the two sensor wires connected to the shunt (normally screw terminals) directly to the shunt PCB (normally soldered to board).
Yes, 2 sensor wires soldered to the board.

>Additionally, you have **two power feeds** going directly from the battery (after positive merged together) to both "+++" and "+VSENSE" terminals ?
I have **1 power feed / positive cable** going from the battery to the shunt and bridge "+++" and "+VSense" on the screw terminals on the shunt board.

stuartpittaway, 2023-02-17:
Thanks, given how low the current reading is, there may be a short/solder bridge on the input terminals.

Can you **disconnect** both of the wires to the shunt "kelvin" connections (small screw terminals) and using a multimeter, test the resistance between the two sensor wires?

Additionally, check the resistance between pins 9 and 10 on the INA228 chip, as those pins are only 0.5mm apart, I suspect there's a very small solder bridge between those two pins - or one of them is not soldered to the PCB correctly.

stuartpittaway, 2023-02-17:
Ok - you might be able to just re-heat the existing board to sort out the solder joints.

red0909, 2023-02-18:
i have the 500 amp 50mv shunt too
no issues

stuartpittaway, 2023-04-24:
@stefanK74 did you get this fixed?

stefanK74, 2023-04-24:
Unfortunately not, I resoldered the INA228 and checked pin9+10, but that didn´t help. I need to look into it one more time..

stefanK74, 2023-05-09:
I assume it´s a soldering issue and put his board on the shelf..
Since INA228s are available again I will populate a new board.
Thanks everyone for your feedback!

closed 2023-05-09 by stefanK74