stuartpittaway/diyBMS-CurrentShunt #17: Shunt resets only to 99%

red0909, 2023-02-18:
the shunt reset to 99% not to 100% looks like a bug.

[Release-2023-01-25-12-26]

stuartpittaway, 2023-02-20:
Do you have "Charge efficiency factor %" set to 99% ?

That will have the effect of subtracting 1% from the 100% on reset.

red0909, 2023-02-20:
yes charge efficency was at 99%

interesting, if the shunt should reset to 100% at full charge it reset to the Charge efficiency factor % value

this was not the case with older firmware

stuartpittaway, 2023-02-23:
The whole SoC calculation is different in this version - the previous one was incorrect.

It doesn't set to the charge efficiency factor % directly.  The SoC is calculated on the amps in vs out, and the factor % is applied to the "in" value to determine the SoC.

red0909, 2023-02-23:
what is the charge efficency factor for, if it has to be set to 100% ?

make no sence for me, ok i will set it to 100%...

stuartpittaway, 2023-02-23:
When you charge any battery, some energy is lost - thats the efficiency.

Lithium cells are very good - hence 99%+ - lead acid less so, probably 90%

red0909, 2023-02-23:
i undestand but in this case if i set 99,5% efficency it wil never reset to 100%... but i have to

stuartpittaway, 2023-02-23:
Ok, set it to 100% then.

deekoowee, 2023-03-27:
I have firmware 40792eb5 and I'm thinking of SoC calculation.
Bank is 14S Lithium and fully charged voltage is set for testing to 58,8V.
Charge efficiency factor is 100%
SoC goes a bit over 100% at 58,8V max, but after a while returns to 100%. 

When discharging, voltage drops for example to 53,14V, SoC changes to 72,34%.
With linear voltage drop calc between 58,8 and 50.4, the SoC at 53V is 30%

Maybe there's something, which I don't underestand.
Any ideas ?

stuartpittaway, 2023-03-27:
Voltage has nothing to do with the SoC calculation !

Its only used to identify when 100% fully charged has been reached - 58.8V in your case.

After that, its down to amp-hour counting.

You can't apply a linear voltage drop calc to Lithium cells, they don't have a linear charge curve.

red0909, 2024-02-23:
any news on this issue?

stuartpittaway, 2024-02-24:
The state of charge does indeed reset to 100% when the conditions are met, however from that 100% value the charge efficiency is removed, typically this introduces the 99% value you see.

red0909, 2024-02-25:
hmm, so i have to live with the 99% on full?
if i set the efficency to 100% then the soc calculation is to high if the battery dont charge full for a week

stuartpittaway, 2024-02-25:
There has been a code change submitted by another use to change the behaviour so that the soc goes to 100%, but that's not been incorporated into the main code base.

red0909, 2024-02-25:
could you please integrate it in the external shunt firmware, if its not to much work?
i dont know how to do it.
thanks

stuartpittaway, 2024-02-27:
This is related to issue https://github.com/stuartpittaway/diyBMSv4ESP32/issues/232

red0909, 2025-01-07:
so can you please fix that issue for the external shunt?