stuartpittaway/diyBMS-CurrentShunt #18: Current Shunt relay logic not activating

cass3825, 2023-04-02:
I cannot seem to trigger the relay. See screenshot for configuration which i believe should be triggering at least three limits.
[image]

stuartpittaway, 2023-04-02:
I'm not sure about using zeros in settings for the relay.

Are you using the latest firmware on the shunt board?

cass3825, 2023-04-02:
Changed zeros to 0.1 and -0.1, same result. Firmware version 40792eb5 dated 25 Jan 2023.

stuartpittaway, 2023-04-03:
Ok, needs to be investigated more thoroughly, but I'm not near a computer for a few days now.

Just for information, what are you running from the shunt relay?

cass3825, 2023-04-03:
Thanks for looking into it! The shunt relay will initiate a generator start, triggered by the under voltage limit.

stuartpittaway, 2023-04-03:
Ok, don't forget the relays on the controller can also do that for you!

cass3825, 2023-04-03:
Controller relays are assigned to Charge Permissive, Discharge Permissive, Non-critical Load Shed & Battery Heater. It seems the more output channels you provide, the more applications we dream up! Lol