stuartpittaway/diyBMS-CurrentShunt #19: Not rally an issue but a request.... couldnt see a request tab!

anthonyevs, 2023-04-23:
Hi Stuart,
I am using the diybms with the original shunt, and its running really well.
I do have a request though.
Is there any chance we can have a tab which could be selected onto our diybms homepage that shows the AH remaining.
I realise e have % capacity remaining, but for me id love to see a AH remaining tab. I dont really use the total ah in and out tabs, but a AH remaining tab for me would be really useful.
If only to see over time the reduction in battery capacity due to losses / wear.
Thanks Stuart.

stuartpittaway, 2023-04-24:
Hi Anthony, at first I thought this was going to be an easy request!

Showing remaining amp-hours based on the fixed capacity provided in the user interface should be straight forward.

You mention you wish to see the capacity due to wear/usage, that sounds quite complicated to calculate.

Any ideas?

anthonyevs, 2023-04-24:
Hi Stuart,
I was thinking, been as the shunt accurately counts in and out the AH into the battery, we could do this.
1. Drain the battery to your personalised setting low cut off point.
2. fill the battery to your personalised settings for the high cut off point.
3. Record the AH taken in by that charge, lets say that is 100AH of energy the battery has taken, so that also equates to 100% SOC. That now represents your starting point going forward, as on this date, your battery at 100% SOC had 100AH capacity.
4. Do the same procedure 1 year in of ownership, and do a drain / charge cycle from your same existing min max personalised settings.. If the battery now only takes 96AH to max charge, this still equates to 100% SOC, but you can see over the 1 yr of ownership and usage, the battery has lost 4 AH of its capacity through wear. 
5. That will show what our battery performance has been over a set time period.
6. Also, i personally like the idea of seeing the AH remaining in a tab, for me it would have a meaning i can relate to and understand (although also SOC remaining is very clear too, but SOC doesnt represent the actual AH remaining)

Hope this makes sense Stuart.