stuartpittaway/diyBMS-CurrentShunt #21: CPL files

BorisvdsHA, 2023-10-18:
CPL files do not seem correct for BMS and shunt. JLPCB refuses to go further. Says positions are missing.

BorisvdsHA, 2023-10-18:
ModuleV490-AllInOne and shunt

stuartpittaway, 2023-10-18:
What is the exact message you are receiving?  Can you upload a screenshot?

BorisvdsHA, 2023-10-18:
I downloaded the gerber zip file, Jl BOM and cpl from this folder. The RAW files.

[diyBMS-CurrentShunt](https://github.com/stuartpittaway/diyBMS-CurrentShunt/tree/master)/[CurrentShuntCircuit](https://github.com/stuartpittaway/diyBMS-CurrentShunt/tree/master/CurrentShuntCircuit)
/export/

BorisvdsHA, 2023-10-18:
[image]

BorisvdsHA, 2023-10-18:
I added the gerber, works perfect. I added BOM and CPL, works perfect. But the next step does not work. Gives this error. I think it is also the case for the BMS 16s.

Nevertheless great work, looking if I can put it together. I have have used easyeda before to test. I have a little experience with esphome an HA. Thanks.

stuartpittaway, 2023-10-19:
That's a new message JLC have introduced recently - you can ignore it, they cannot fit those parts so they are excluded from the BOM

closed 2023-10-19 by stuartpittaway

BorisvdsHA, 2023-10-19:
OK thanks.